KillerInk/FreeDcam #126: Bayer to DNG conversion gives no files or 0 bytes DNGs

viktorkrasovsky, 2020-08-26:
Phone Model: Xiaomi Redmi note 6 Pro
---------------

FreeDcam Version: 4.3.3
-----------------

Problem Description: I was using Freedcam 4.2 and raw to DNG conversion worked with no essential problem. Later, when the 4.3.3 update came and I started the App the camera was in JPG mode. I wanted to switch back and found only bayer mode. Ok, I was thinking I need to setup RAW 2 DNG configuration again but that did not help. I tried to convert the bayer file to DNG with the FreedcamViewer and nothing happens in any DNG output mode. A while ago I had a backup for dngprofiles.xml and restored to the FreedCam/config folder and tried again without success. Ok, I started the FreedCam app, the DNG option appeared, however the DNGs it takes now take 0 bytes. I rolled back to the 4.2 version and DNG conversion works without issue. Any Idea how could I fix the problem? Could not find any log files.
--------------------

KillerInk, 2020-08-26:
i missed to add the saf changes for the dngconverter.
also the path changed where the dngprofile should get stored.
its now in /sdcard/Android/data/troop.com.freedcam/files/

that changes are needed to get rdy for android11 and scoped storage

viktorkrasovsky, 2020-08-26:
@KillerInk thanks, I copyed the old profile into the new path and the freedcam can shoot DNGs again!

closed 2020-08-26 by viktorkrasovsky