KillerInk/FreeDcam #133: Huawei P9+ Mono sensor DNG issue

sarurururu, 2020-11-17:
Hi, in the recent releases of FreeDcam the DNGs saved will become messed up like this:
[image]

It will be solved after switching the "Dual Camera" under settings to color, then back to mono. This problem comes back if I leave the app (even without killing it), e.g. pressing the home button.

Thanks for such a great app!

KillerInk, 2020-11-20:
hi the problem is caused due settings are not set to the camera. 
its fixed in meantime and with next release its working again

sarurururu, 2020-11-23:
Thanks, I've just updated to 4.3.16 and it crashes upon starting. It persists after a reboot and re-installation.

KillerInk, 2020-11-23:
yeah added something stupid. 4.3.17 work again

sarurururu, 2020-11-24:
The app starts again but the same mono DNG issue is still there...