KillerInk/FreeDcam #134: crash in video recording

DoctorSubtilis, 2020-11-19:
I don't know why but every time I try to record a video, the app crash, or better goes away (from full screen, and becomes a window in background).
I have a Motorola E6 Plus, I tried changing some video settings (max video size: from 0 to 50 mb, f.e.), but nothing to do.

KillerInk, 2020-11-20:
hi pls try to set max recording time and max file size to 0. then mediarecorder ignores it.
some devices dont like it and crash in that case. 
have to add more feedback to users if something while preparing fails.

DoctorSubtilis, 2020-11-20:
Thank you. Done, but unsuccessfully.  
I tried to disable audio recording as well, but likewise unsuccessfully...

KillerInk, 2020-12-03:
Feel free to open issue again when you provide more informations

DoctorSubtilis, 2020-12-03:
> Feel free to open issue again when you provide more informations
Yes, could you explain how can I provide you the log? Thank you.

KillerInk, 2020-12-03:
https://github.com/KillerInk/FreeDcam/wiki/How-to-Log

DoctorSubtilis, 2020-12-22:
Here the log: 
[log.txt](https://github.com/KillerInk/FreeDcam/files/5731261/log.txt)

DoctorSubtilis, 2020-12-28:
After sending the log, couldn't you religios-open this thread? Thanks!

DoctorSubtilis, 2020-12-30:
Well, today, after uninstalling and reinstalling the app,  I managed to make a video without crashing.

closed 2021-07-11 by KillerInk