KillerInk/FreeDcam #138: 90 Degree rotation on Action Cam

metalx1000, 2020-12-12:
**Describe the bug**
The Motorola G Stylus has 3 cameras, one of which is an "Action Camera" which is a wide angle camera for video.  The way the camera is positioned on this phone, you hold the camera in portrait mode and it takes video in Landscape.  With FreeDcam, the video is not rotating correctly.  There is an option to rotate 180 degrees, but this phone needs the image rotated 90 degrees.  Other wise the video is always sideways on the screen while recording.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to camera 'Back 3' and image is turned 90 degrees

**Expected behavior**
Image should be turned.

**Screenshots**
https://photos.app.goo.gl/FagRQXQL7kTZ9UiJ7

**Smartphone (please complete the following information):**
 - Device: Moto G Stylus (sofiap)
 - OS: Android 10
 - Version  4.3.17

KillerInk, 2020-12-13:
hi. the preview itself looks ok to me. its not streched or compressed.
i expect with rotating it about 90° it get compressed.
how does the 3rd cam act on other camera apps?

metalx1000, 2020-12-13:
So the issue is that the camera is turned 90° physically on the phone.
So the image needs to be turned in the software, other wise when I move the camera left and right, the image on the screen moves up and down, and when I move up and down the video on the screen moves left and right.

I hope this video helps explain it.
https://photos.app.goo.gl/BiNPam8msrExPeNu5

FreeDcam is the only Open Source Camera app I've found that even detects this camera, but sadly it's pretty much impossible to film anything this way.

KillerInk, 2020-12-13:
one more thing. have the recorded file the same problem or is the orientation normal?
will change the code that you can rotatet it 0,90,180,270

metalx1000, 2020-12-13:
This video seems to be saving properly.  By that I mean that if I hold the camera on portrait mode it records in landscape mode, which is how it is suppose to be with this phone.  I hope that makes sense, I understand it's a little confusing.  I've not seen another camera on a phone like this.  But as far as I can tell, if we can get the preview screen to rotate, the video is saving correctly.

Only other thing I see is that the preview while using this camera is a little jerky, but the recording looks fine.
I can upload some more test videos if that would help explain things better.

KillerInk, 2020-12-13:
https://github.com/KillerInk/FreeDcam/releases/tag/4.3.20

hope it works as it should. here it rotate in correct order. but i expect that the preview is now a bit streched^^

metalx1000, 2020-12-13:
Wow, you are really quick on your responses.  I appreciate it so much.
Unfortunately with that apk, camera "Back 3" doesn't work at all any more.
It shows up in the list, but when I choose it, the preview from the last camera selected freezes and Camera 3 never shows up.
:(

KillerInk, 2020-12-13:
yeah made a mistake, next try^^
https://github.com/KillerInk/FreeDcam/releases/tag/4.3.21

metalx1000, 2020-12-13:
We're 90% there.  The video is turned the correct way, and the saved video looks correct (both orientation and aspect), the only issue now is the preview of the video while recording is squished.  So, it's usable but not ideal.  I'm heading how with my kids to the park now.  I'll try and use the camera while we are there and let you know if everything else is working correctly.

Thanks again for your quick response.

metalx1000, 2020-12-14:
So, I recorded this video to show where we're currently at:
https://photos.app.goo.gl/icz6mQMVXC1jRN5g7

I hope doing something like the letter boxing is possible.
I know it's hard to support all the different devices out there and the Android System doesn't make things easy.  So I again want to thank you for all your help.

KillerInk, 2020-12-14:
there is something odd with the returned mount angle from the action cam.
my solution would be to add a new option wich can switch the aspect ratio.

KillerInk, 2020-12-14:
https://github.com/KillerInk/FreeDcam/releases/tag/4.3.22
next try

metalx1000, 2020-12-14:
YES!
I've only one one quick test so far, but everything seems to be work.
Thank you so much!

closed 2020-12-16 by metalx1000