KillerInk/FreeDcam #139: No Video possible on Huawei P20 Pro (again)

beniroquai, 2020-12-14:
**Describe the bug**
After a fresh reinstall from the Play store, I have problems capturing a video. There are multiple problems it seems, first being the missing permission (Huawei only problem? See Screenshot #1). The Logcat gives:

```
2020-12-14 08:12:21.597 10603-10659/? E/AndroidRuntime: FATAL EXCEPTION: ModuleHandlerAbstract
    Process: troop.com.freedcam, PID: 10603
    java.lang.SecurityException: Permission Denial: writing com.android.providers.media.MediaProvider uri content://media/external/video/media from pid=10603, uid=10195 requires android.permission.WRITE_EXTERNAL_STORAGE, or grantUriPermission()
        at android.os.Parcel.createException(Parcel.java:1953)
        at android.os.Parcel.readException(Parcel.java:1921)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:183)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:135)
        at android.content.ContentProviderProxy.insert(ContentProviderNative.java:476)
        at android.content.ContentResolver.insert(ContentResolver.java:1611)
        at freed.file.MediaStoreController.addMovie(MediaStoreController.java:78)
        at freed.file.FileListController.getNewMovieFileHolder(FileListController.java:335)
        at freed.cam.apis.basecamera.record.VideoRecorder.setRecorderFilePath(VideoRecorder.java:267)
        at freed.cam.apis.basecamera.record.VideoRecorder.prepare(VideoRecorder.java:180)
        at freed.cam.apis.camera2.modules.VideoModuleApi2.startPreviewVideo(VideoModuleApi2.java:388)
        at freed.cam.apis.camera2.modules.VideoModuleApi2.startRecording(VideoModuleApi2.java:179)
        at freed.cam.apis.camera2.modules.VideoModuleApi2.lambda$startStopRecording$0$VideoModuleApi2(VideoModuleApi2.java:108)
        at freed.cam.apis.camera2.modules.-$$Lambda$VideoModuleApi2$X55bREiLuDD0ZRON-qS0V1YzYxs.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:907)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:216)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```

After manually granting permission, The following error occurs:

```
2020-12-14 08:14:08.258 3872-8255/? I/ctaifs: <FreeDCam>[FreeDCam][troop.com.freedcam]:[openFile] 打开文件
2020-12-14 08:14:08.268 11288-11343/? W/System.err: java.io.FileNotFoundException: Can't access /MOV_2020_12_14_08_14_08.mp4
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at android.database.DatabaseUtils.readExceptionWithFileNotFoundExceptionFromParcel(DatabaseUtils.java:144)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at android.content.ContentProviderProxy.openAssetFile(ContentProviderNative.java:625)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at android.content.ContentResolver.openAssetFileDescriptor(ContentResolver.java:1331)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at android.content.ContentResolver.openFileDescriptor(ContentResolver.java:1166)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at android.content.ContentResolver.openFileDescriptor(ContentResolver.java:1120)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at freed.file.holder.UriHolder.getParcelFileDescriptor(UriHolder.java:108)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at freed.file.holder.BaseHolder.setToMediaRecorder(BaseHolder.java:127)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at freed.cam.apis.basecamera.record.VideoRecorder.setRecorderFilePath(VideoRecorder.java:269)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at freed.cam.apis.basecamera.record.VideoRecorder.prepare(VideoRecorder.java:180)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.VideoModuleApi2.startPreviewVideo(VideoModuleApi2.java:388)
2020-12-14 08:14:08.269 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.VideoModuleApi2.startRecording(VideoModuleApi2.java:179)
2020-12-14 08:14:08.270 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.VideoModuleApi2.lambda$startStopRecording$0$VideoModuleApi2(VideoModuleApi2.java:108)
2020-12-14 08:14:08.270 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.-$$Lambda$VideoModuleApi2$X55bREiLuDD0ZRON-qS0V1YzYxs.run(Unknown Source:2)
2020-12-14 08:14:08.270 11288-11343/? W/System.err:     at android.os.Handler.handleCallback(Handler.java:907)
2020-12-14 08:14:08.270 11288-11343/? W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:105)
2020-12-14 08:14:08.270 11288-11343/? W/System.err:     at android.os.Looper.loop(Looper.java:216)
2020-12-14 08:14:08.270 11288-11343/? W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:65)
2020-12-14 08:14:08.271 11288-11343/? W/System.err: java.io.IOException: No valid output file
2020-12-14 08:14:08.271 11288-11343/? W/System.err:     at android.media.MediaRecorder.prepare(MediaRecorder.java:1035)
2020-12-14 08:14:08.271 11288-11343/? W/System.err:     at freed.cam.apis.basecamera.record.VideoRecorder.prepare(VideoRecorder.java:257)
2020-12-14 08:14:08.271 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.VideoModuleApi2.startPreviewVideo(VideoModuleApi2.java:388)
2020-12-14 08:14:08.271 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.VideoModuleApi2.startRecording(VideoModuleApi2.java:179)
2020-12-14 08:14:08.271 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.VideoModuleApi2.lambda$startStopRecording$0$VideoModuleApi2(VideoModuleApi2.java:108)
2020-12-14 08:14:08.272 11288-11343/? W/System.err:     at freed.cam.apis.camera2.modules.-$$Lambda$VideoModuleApi2$X55bREiLuDD0ZRON-qS0V1YzYxs.run(Unknown Source:2)
2020-12-14 08:14:08.272 11288-11343/? W/System.err:     at android.os.Handler.handleCallback(Handler.java:907)
2020-12-14 08:14:08.272 11288-11343/? W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:105)
2020-12-14 08:14:08.272 11288-11343/? W/System.err:     at android.os.Looper.loop(Looper.java:216)
2020-12-14 08:14:08.272 11288-11343/? W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:65)
2020-12-14 08:14:08.272 11288-11343/? D/ModuleAbstract: work started
2020-12-14 08:14:08.272 11288-11343/? D/ShutterButton: switchBackground:video_recording_stop
2020-12-14 08:14:08.277 774-11379/? I/HwaStatInfo: [3c6ca2b_64I] algotype is 0, videoRatio is 1.000000, ratio is 2.500000, faceNum is 0
````

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest Freedcam from Playstore
2 Manually Grant Storage Permission from EMUI
3.Start video capturing
4. No file will be created

**Expected behavior**
Recording a video should start. 

**Screenshots**

## Screenshot ```#1``` 

[image]

## Screenshot ```#1``` 

![image](https://user-images.githubusercontent.com/4345528/102051438-dcbace80-3de4-11eb-8b99-6c02d486460b.png
[image]
)

**Smartphone (please complete the following information):**
 - Device: [e.g. Xiaomi PocoF2]
 - OS: [e.g. Android9]
 - Version [e.g. 4.3.8]

**Additional context**
EMUI has recently been updated; Perhaps a problem? The the cellSTORM APP still runs without a problem ;)

I merged into recent Master branch and will have a look what's going on there :)

KillerInk, 2020-12-14:
write permission shouldnt needed with mediastore and >= api28

https://developer.android.com/training/data-storage/shared/media

thats also why freedcam nolonger ask for write permission.

happen it only with movies or images too?

beniroquai, 2020-12-14:
Taking images (after recompiling from source and not setting permission manually) gives:

```
020-12-14 11:55:24.180 24091-24275/troop.com.freedcam E/AndroidRuntime: FATAL EXCEPTION: pool-3-thread-1
    Process: troop.com.freedcam, PID: 24091
    java.lang.SecurityException: Permission Denial: writing com.android.providers.media.MediaProvider uri content://media/external/images/media from pid=24091, uid=10196 requires android.permission.WRITE_EXTERNAL_STORAGE, or grantUriPermission()
        at android.os.Parcel.createException(Parcel.java:1953)
        at android.os.Parcel.readException(Parcel.java:1921)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:183)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:135)
        at android.content.ContentProviderProxy.insert(ContentProviderNative.java:476)
        at android.content.ContentResolver.insert(ContentResolver.java:1611)
        at freed.file.MediaStoreController.addImg(MediaStoreController.java:57)
        at freed.file.FileListController.getNewImgFileHolder(FileListController.java:344)
        at freed.image.ImageSaveTask.saveJpeg(ImageSaveTask.java:234)
        at freed.image.ImageSaveTask.process(ImageSaveTask.java:172)
        at freed.image.ImageTask.run(ImageTask.java:21)
        at java.util.concurrent.ThreadPoolExecutor.processTask(ThreadPoolExecutor.java:1187)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:784)
2020-12-14 11:55:24.192 24091-24100/troop.com.freedcam W/System: A resource failed to call destroy. 
2020-12-14 11:55:24.193 24091-24100/troop.com.freedcam D/BackgroundHandlerThread: stopBackgroundThreadCaptureSessionHandler
2020-12-14 11:55:24.194 24091-24100/troop.com.freedcam W/System: A resource failed to call release. 
2020-12-14 11:55:24.194 24091-24100/troop.com.freedcam I/chatty: uid=10196(troop.com.freedcam) FinalizerDaemon expire 1 line
2020-12-14 11:55:24.194 24091-24100/troop.com.freedcam W/System: A resource failed to call release. 
2020-12-14 11:55:24.220 24091-24091/troop.com.freedcam D/LocationManager: stop location
2020-12-14 11:55:24.224 24091-24275/troop.com.freedcam I/Process: Sending signal. PID: 24091 SIG: 9
```

After setting permission in settings, it creates the following output (no error, but also no picture):
```
2020-12-14 12:00:50.812 24494-24494/troop.com.freedcam V/AudioManager: playSoundEffect   effectType: 0
2020-12-14 12:00:50.812 24494-24494/troop.com.freedcam V/AudioManager: querySoundEffectsEnabled...
2020-12-14 12:00:50.812 24494-24494/troop.com.freedcam D/PictureModuleApi2: startWork: start new progress
2020-12-14 12:00:50.813 24494-25071/troop.com.freedcam D/PictureModuleApi2: Jpeg
2020-12-14 12:00:50.813 24494-25071/troop.com.freedcam D/PictureModuleApi2: mState:STATE_WAIT_FOR_PRECAPTURE
2020-12-14 12:00:50.813 24494-25071/troop.com.freedcam D/CameraValuesChangedCaptureCallback: rest ae af lock
2020-12-14 12:00:50.813 24494-25071/troop.com.freedcam D/PictureModuleApi2: Start AE Precapture
2020-12-14 12:00:50.813 24494-25071/troop.com.freedcam D/CaptureSessionHandler: Set :android.control.aePrecaptureTrigger to 1
2020-12-14 12:00:50.813 24494-25071/troop.com.freedcam D/CaptureSessionHandler: Set :android.control.afTrigger to 1
2020-12-14 12:00:50.813 24494-25071/troop.com.freedcam D/CaptureSessionHandler: capture
2020-12-14 12:00:50.842 24494-25071/troop.com.freedcam D/CameraValuesChangedCaptureCallback: ae locked: true af locked: false 1188
2020-12-14 12:00:50.842 24494-25071/troop.com.freedcam D/PictureModuleApi2: ae locked: true af locked: false
2020-12-14 12:00:50.876 24494-25071/troop.com.freedcam D/CameraValuesChangedCaptureCallback: ae locked: true af locked: false 1188
2020-12-14 12:00:50.876 24494-25071/troop.com.freedcam D/PictureModuleApi2: ae locked: true af locked: false
2020-12-14 12:00:50.910 24494-25072/troop.com.freedcam D/CameraValuesChangedCaptureCallback: ae locked: true af locked: false 1189
2020-12-14 12:00:50.910 24494-25072/troop.com.freedcam D/PictureModuleApi2: ae locked: true af locked: false
2020-12-14 12:00:50.944 24494-25071/troop.com.freedcam D/CameraValuesChangedCaptureCallback: ae locked: true af locked: true 1188
2020-12-14 12:00:50.944 24494-25071/troop.com.freedcam D/PictureModuleApi2: ae locked: true af locked: true
2020-12-14 12:00:50.945 24494-25071/troop.com.freedcam D/CameraValuesChangedCaptureCallback: clear wait for ae af lock
2020-12-14 12:00:50.945 24494-25071/troop.com.freedcam D/PictureModuleApi2: mState:STATE_PICTURE_TAKEN
2020-12-14 12:00:50.945 24494-25071/troop.com.freedcam D/PictureModuleApi2: ########### captureStillPicture ###########
2020-12-14 12:00:50.948 24494-25071/troop.com.freedcam D/CaptureSessionHandler: Set :android.jpeg.orientation to 90
2020-12-14 12:00:50.948 24494-25071/troop.com.freedcam D/ModuleAbstract: work started
2020-12-14 12:00:50.948 24494-25071/troop.com.freedcam D/ShutterButton: switchBackground:image_capture_start
2020-12-14 12:00:50.948 24494-25071/troop.com.freedcam D/PictureModuleApi2: StartStillCapture
2020-12-14 12:00:50.949 24494-25071/troop.com.freedcam D/CaptureSessionHandler: StopRepeatingCaptureSession
2020-12-14 12:00:50.949 24494-25071/troop.com.freedcam D/CaptureSessionHandler: StartImageCapture
[... cut ...]
2020-12-14 12:01:06.147 24494-24494/troop.com.freedcam D/OverScrollerOptimization: fling time is flingTime = 2.406646330383331 velocity = -3109
2020-12-14 12:01:06.147 24494-24494/troop.com.freedcam I/chatty: uid=10196(troop.com.freedcam) identical 2 lines
2020-12-14 12:01:06.147 24494-24494/troop.com.freedcam D/OverScrollerOptimization: fling time is flingTime = 2.406646330383331 velocity = -3109
2020-12-14 12:01:07.331 24494-24494/troop.com.freedcam W/HiTouch_PressGestureDetector: Touch pointer move a lot. The moving distance of X is:99.01221, limit is:60The moving distance of Y is:22.758545, limit is:60
```

KillerInk, 2020-12-14:
its the same problem. it have something todo with the contentprovider freedcam use

```
219/troop.com.freedcam W/System.err: java.io.FileNotFoundException: Can't access /2020_12_14_12_00_50.jpg
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at android.database.DatabaseUtils.readExceptionWithFileNotFoundExceptionFromParcel(DatabaseUtils.java:144)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at android.content.ContentProviderProxy.openAssetFile(ContentProviderNative.java:625)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at android.content.ContentResolver.openAssetFileDescriptor(ContentResolver.java:1331)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at android.content.ContentResolver.openOutputStream(ContentResolver.java:1073)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at android.content.ContentResolver.openOutputStream(ContentResolver.java:1049)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at freed.file.holder.UriHolder.getOutputStream(UriHolder.java:108)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at freed.image.ImageSaveTask.saveJpeg(ImageSaveTask.java:236)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at freed.image.ImageSaveTask.process(ImageSaveTask.java:172)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at freed.image.ImageTask.run(ImageTask.java:21)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at java.util.concurrent.ThreadPoolExecutor.processTask(ThreadPoolExecutor.java:1187)
2020-12-14 12:00:51.353 24494-25219/troop.com.freedcam W/System.err:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)
2020-12-14 12:00:51.354 24494-25219/troop.com.freedcam W/System.err:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
2020-12-14 12:00:51.354 24494-25219/troop.com.freedcam W/System.err:     at java.lang.Thread.run(Thread.java:784)
```

KillerInk, 2020-12-14:
think i have to blame me^^

https://github.com/KillerInk/FreeDcam/blob/88d1ad6fdd4558e11bfd4589e743f7ccd8c06645/app/src/main/java/freed/file/FileListController.java#L65

may get replace with

`public static boolean needStorageAccessFrameWork = Build.VERSION.SDK_INT > Build.VERSION_CODES.P; `

beniroquai, 2020-12-14:
Yes, fixed! Awesome job - again!

closed 2020-12-14 by beniroquai