KillerInk/FreeDcam #143: Add a trigger event for finished capture

Wandang, 2021-01-01:
**Describe the bug**
Using this app in conjunction with [osmAnd](https://osmand.net/) does not work. Taking a picture for a location sends an intent and FreeDCam correctly offers itself to take the picture. After capturing via the pressing the big button the context does not switch back to osmAnd.

**To Reproduce**
Steps to reproduce the behavior (assuming that this behavior is independent from osmAnd itself):
1. Go to any third party app that wants to use the camera
2. Choose FreeDCam to capture the picture
3. Press the capture button

**Expected behavior**
After pressing the capture button I would expect to see the picture (granted, I can just swipe to the right to see it) and then be presented a choice to keep or to discard the picture. After that the context should switch back to osmAnd which takes the returned picture from FreeDCam to be used further.

**Smartphone (please complete the following information):**
 - Device: OnePlus 6 (enchilada)
 - OS: Android 10
 - Version LineageOS 17.1

KillerInk, 2021-01-17:
hi
im disabling it. till someone have the time and motivations to fix up the picmodules and handel images format send due that intent.

Wandang, 2021-01-17:
Unfortunately my android/java skills are to basic to help. Thanks for the headsup and removing the confusion for now! The app itself is quite wondeful