KillerInk/FreeDcam #147: micro sensor not available

pp058, 2021-02-15:
micro sensor or camera not available in back not showing 
camera 2api 
Android 10 
moto g9

KillerInk, 2021-02-16:
I don't know what I'd get used for that cam. You have to provide more informations

pp058, 2021-02-18:
> I don't know what ID get used for that cam. You have to provide more informations

it work fine in 4.3.3 version but not in 4.17 
what kind of information you need tell ??

KillerInk, 2021-02-20:
it is as it is. on that version some devices didnt find a camera. so it got reverted.
you have to wait till freedcam get multicamera api added. but feel free to contribute, its open source

closed 2021-02-20 by KillerInk

pp058, 2021-02-20:
I know its open source but i didn't have the knowledge about Android app building but 
i can be your tester for sure 😄😄