KillerInk/FreeDcam #148: interval doesn't behave as it should

yzak4815, 2021-02-19:
interval doesn't behave as it should
intervalometer fails. When the shutter interval is programmed in 30s with the interval duration in 1 minute, it should take 2 or at most 3 photos, but the program takes hundreds of photos and when counting down the time is not precise, it goes from 30 to 0 in 5 seconds and takes photos every 1-2 seconds. no matter what parameters are set in the interval options, it doesn't behave as it should.

**Expected behavior**
The intervalomer should behave according to its programming, if you set the inteval shutter speed to 15s, it should take a photo every 15 seconds until the end of the interval period.

**Smartphone:
 - Device: [Motorola Moto G8]
 - OS: [Android 10]

KillerInk, 2021-02-28:
here it runs twice fast, will check whats going on

KillerInk, 2021-08-03:
should be fixed with [4.3.51](https://github.com/KillerInk/FreeDcam/releases/tag/4.3.51)

closed 2021-08-03 by KillerInk