KillerInk/FreeDcam #153: front camera reset or many little issue

fob014, 2021-03-12:
(1-:) on front camera settings like scene mode, manual exposure, saturation, sharpening get rest after relaunch app
(2-:) on back & front  both there is no option showing to adjust brightness , contras . 
camera2 api enable 
Android 10 
not device specific

KillerInk, 2021-03-13:
1. when in remember correct freedcam dont apply the settings from manual to front cam because it crash them. but front cam have more problems. preview gets upside down when rotating the device to portrait. myself dont use the front cam so it get no love.
2. camera2 by default dont support it. only some qcom socs have a hidden key for it.
3. ae bracket works mostly only on full devices with per frame control.

fob014, 2021-03-13:
> 1. when in remember correct freedcam dont apply the settings from manual to front cam because it crash them. but front cam have more problems. preview gets upside down when rotating the device to [...]

hmm

fob014, 2021-03-13:
> 1. when in remember correct freedcam dont apply the settings from manual to front cam because it crash them. but front cam have more problems. preview gets upside down when rotating the device to [...]

in stock camera iso goes up to 3200 & exposure also get up to 32 sec long but in freedcam iso only 1600 & exposure only 1/8 
camera2 api

KillerInk, 2021-03-13:
i dont know. oems do alot stupid things. maybe your device is hardcoded with that limits in stock cam.
what freedcam use is what get reported from the camera.
but you can extend it manual inside  /Android/data/troop.com.freedcam/files/freed_config.xml

fob014, 2021-03-14:
> i dont know. oems do alot stupid things. maybe your device is hardcoded with that limits in stock cam.
> what freedcam use is what get reported from the camera.
> but you can extend it manual inside /Android/data/troop.com.freedcam/files/freed_config.xml

so editing xml file for  long exposure but  not for iso 
i add 3200 iso and even set 3200 iso in camera app iso manual but i see  no change in viewfinder  
but thanks its work 👍 for exposure

KillerInk, 2021-03-15:
check the metadata from a captured image. not all devices show what you finaly get.
google removed it due their increasing api requirements and i decided to keep backward compatibility.

fob014, 2021-03-16:
yes i checked metdata its also show 1600 not 3200 
look like editing xml didn't work for iso but is does work for exposure

fob014, 2021-03-16:
> check the metadata from a captured image. not all devices show what you finaly get.
> google removed it due their increasing api requirements and i decided to keep backward compatibility.

yes i checked metdata its also show 1600 not 3200 
look like editing xml didn't work for iso but is does work for exposure