KillerInk/FreeDcam #156: Crash of v4.3.39, v4.3.38 when starting

Codereamp, 2021-03-16:
**Describe the bug**
Freedcam just installed can not start, the black screen is visible for a second then the launcher again.

**To Reproduce**
- Install v4.3.39 or v4.3.38
- Start the app
- See a black screen for a second, then the launcher again. 

**Expected behavior**
The working window should be visible. Actually sometimes the app started successfully, both for v4.3.39 and v4.3.38, but in minority of the cases. Previous version in the release line (4.3.22debug) successfully starts several times in a row. 

**Screenshots**

**Smartphone (please complete the following information):**
 - Device: Meizu M6 Note
 - OS: Android7, Flyme 7.1.1.0
 - Version: 4.3.38, 4.3.39

**Additional context**
Eventlog contains a crash error (checked two times, the text is identical).
03-16 16:15:24.483 I/am_crash( 1718): [4727,0,troop.com.freedcam,949534276,android.view.ViewRootImpl$CalledFromWrongThreadException,Only the original thread that created a view hierarchy can touch its views.,ViewRootImpl.java,7048]

Codereamp, 2021-03-16:
the trace from the dropbox of the Crash

android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
	at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:7048)
	at android.view.ViewRootImpl.invalidateChildInParent(ViewRootImpl.java:1102)
	at android.view.ViewGroup.invalidateChild(ViewGroup.java:5205)
	at android.view.View.invalidateInternal(View.java:13761)
	at android.view.View.invalidate(View.java:13725)
	at android.view.View.invalidateParentIfNeeded(View.java:13914)
	at android.view.TextureView.setTransform(TextureView.java:527)
	at freed.cam.previewpostprocessing.NormalPreview.setRotation(NormalPreview.java:127)
	at freed.cam.previewpostprocessing.PreviewController.setRotation(PreviewController.java:158)
	at freed.cam.apis.camera1.modules.PictureModule.createPreview(PictureModule.java:205)
	at freed.cam.apis.camera1.modules.PictureModule.InitModule(PictureModule.java:148)
	at freed.cam.apis.basecamera.modules.ModuleHandlerAbstract.setModule(ModuleHandlerAbstract.java:90)
	at freed.cam.apis.camera1.parameters.ParametersHandler.initParameters(ParametersHandler.java:480)
	at freed.cam.apis.camera1.parameters.ParametersHandler.LoadParametersFromCamera(ParametersHandler.java:132)
	at freed.cam.apis.camera1.Camera1.initCamera(Camera1.java:67)
	at freed.cam.apis.basecamera.CameraThreadHandler.handleMessage(CameraThreadHandler.java:182)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.os.HandlerThread.run(HandlerThread.java:61)

KillerInk, 2021-08-03:
should be fixed with [4.3.51](https://github.com/KillerInk/FreeDcam/releases/tag/4.3.51)

Codereamp, 2021-08-10:
Thanks for fixing, unfortunately the issue is not gone, I'm posting the fragment from the Crash (including the version just to be sure). This time I didn't met a single case when the app window is visible. Every attempt to run led to an immediate exit. Tried to uninstall/reinstall, the same

Process: troop.com.freedcam
Flags: 0x3898be44
Package: troop.com.freedcam v251 (4.3.51)
Foreground: Yes
Build: Meizu/meizu_M6Note/M6Note:7.1.2/N2G47H/m1721.Flyme_7.0.1545529059:user/release-keys

android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
	at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:7048)
	at android.view.ViewRootImpl.invalidateChildInParent(ViewRootImpl.java:1102)
	at android.view.ViewGroup.invalidateChild(ViewGroup.java:5205)
	at android.view.View.invalidateInternal(View.java:13761)
	at android.view.View.invalidate(View.java:13725)
	at android.view.View.invalidateParentIfNeeded(View.java:13914)
	at android.view.TextureView.setTransform(TextureView.java:527)
	at freed.cam.previewpostprocessing.NormalPreview.setRotation(NormalPreview.java:122)
	at freed.cam.previewpostprocessing.PreviewController.setRotation(PreviewController.java:247)
	at freed.cam.apis.camera1.modules.PictureModule.createPreview(PictureModule.java:208)
	at freed.cam.apis.camera1.modules.PictureModule.InitModule(PictureModule.java:151)
	at freed.cam.apis.basecamera.modules.ModuleHandlerAbstract.setModule(ModuleHandlerAbstract.java:85)
	at freed.cam.apis.camera1.parameters.ParametersHandler.initParameters(ParametersHandler.java:481)
	at freed.cam.apis.camera1.parameters.ParametersHandler.LoadParametersFromCamera(ParametersHandler.java:132)
	at freed.cam.apis.camera1.Camera1.initCamera(Camera1.java:69)
	at freed.cam.apis.basecamera.CameraThreadHandler.handleMessage(CameraThreadHandler.java:168)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.os.HandlerThread.run(HandlerThread.java:61)