KillerInk/FreeDcam #158: Feature Request: Timelapse (Fire the Timer Repeatedly)

shuangye, 2021-04-19:
Will you add timelapse support?
You can fire the timer repeatedly, allowing to save JPG/RAW for post processing on a computer. Just like most DSLRs will do.
Besides, you can record a video at the same time for quick preview.

To save battery life, you can allow the user to dim the screen, or even turn off the screen.

KillerInk, 2021-04-19:
Hi
FreeDcam have that feature already. its called interval for pictures.
for video you must create a new mediaprofile.

but they are not working together. and there are no plans to support it due the fact image capture gets limited while recording.

Turn off the screen wont work because we cannot unlock the device once it got turned off.
Dimming can get done manual in the navbar

greets

shuangye, 2021-04-19:
Hi,
Thank you for your reply.
I see the interval button, but seems it can only capture few pictures. To capture timelapse, the total pictures number may be as many as 1000, and the interval may vary from 1s (cloud moving) to several minutes (plants growing).
Turning off the screen does not mean locking the device. You can prevent the device from locking/sleeping (e.g., what video players do), just turn off the screen. I know an App names "Framelapse" has such functionality, but it can only record timelapse videos, without separate pictures, which is essential for professional users to do post processing.

shuangye, 2021-04-20:
If Android does not support capture still images and videos at the same time, maybe a wroundaround could be:
Capture JPG files and encode it into a video using a encoder. Hardware encoder is preferred. If no HW encoder available, or at least SW encoder (e.g., ffmpeg) can be used.

KillerInk, 2021-04-21:
> I see the interval button, but seems it can only capture few pictures. To capture timelapse, the total pictures number may be as many as 1000, and the interval may vary from 1s (cloud moving) to several minutes (plants growing).

you can navigate to settings with a swip from left to right on the preview.
there is a section "Interval Mode Settings"
there you can set the time after the shutter should trigger and how long the interval duration should be.

in the settings you find also the videoprofile editor. there you can create custimzd video profiles^^
like 8k recording(if supported by the device), 10bit8k. or 4k timelapse etc

> Turning off the screen does not imply locking the device. You can prevent the device from locking/sleeping (e.g., what video players do), just turn off the screen (or fill it with black)

Thats then dimming. some devices turn the screen off at lowest lvl.
but drawing a black overlay is possible

>If Android does not support capturing still images and videos at the same time, maybe a wroundaround could be:

camera2 support it. but its then limited to specific image sizes and format i think. 
i only remember that i read about a limitation in android dev docs.

shuangye, 2021-05-17:
I find an interesting workaround:

- Use any camera App, e.g., Android built-in Camera App, even pro mode to get RAW images.
- Install an Auto Clicker to fire the capture. The interval can be ms to hours.
- Install "Proximity Service" https://github.com/ssaqua/ProximityService Cover the Proximity sensor to turn off the screen.