KillerInk/FreeDcam #163: there is a delay in capturing the photo only in back not in front

fob014, 2021-08-01:
delay of when i touch the shutter when it actually start capturing the photo only in back camera not in front 

camera 2 api 
android 10 , 11 both 
camera 2 api level 3

KillerInk, 2021-08-01:
freedcam waits till the exposuretrigger and focus is done or a timout trigger. then it starts a capture.
due that you front cam dont behave like back cam, i expect you front is fixed focus.
to confirm thats it is the focus try to set focus to manual and capture. if the delay is gon,e something is wrong with your autofocus.

fob014, 2021-08-01:
i think no because 
#1  after relaunching the app when first time i capture the photo that delay is not there  but after that it appear 
#2 even in maunal focus , shutter speed, iso it still have that delay 
its must be a bug
i didn't not have this  delay in other third party app 

see this video ( Focus on my finger touch)  first time after launching the app  that delay was not here  after 

https://user-images.githubusercontent.com/80119337/127772448-0ff5fab2-4759-4010-b343-9784b5a218d7.mp4

fob014, 2021-08-01:
https://user-images.githubusercontent.com/80119337/127772725-e45b3513-8334-4172-9722-3a05e5850723.mp4

fob014, 2021-08-01:
https://user-images.githubusercontent.com/80119337/127772787-be8ac322-4781-4003-a7c7-3ddca7cc397d.mp4

KillerInk, 2021-08-02:
i dont think thats a bug. the shutter button change when a image capture get started, and when done.
so on click doesnt mean that there is directly a image capture started.
currently i added a timetrace wich shows the time from button click till the image capture starts.
the measured times are between 5ms - 200ms depening on light condition and how long it took to find focus.

back and front cam are using the same code. i cant repoduce your problem. from my side it looks ok.

if you wanna take a series of shots use the burst slider to set how many images it should take

[image]

fob014, 2021-08-25:
> i dont think thats a bug. the shutter button change when a image capture get started, and when done.
> so on click doesnt mean that there is directly a image capture started.
> currently i added a [...]

than can you add an option to disable or enable that 5ms or 200ms delay