KillerInk/FreeDcam #164: FIX viewfinder image messed up in portrait view #rotation #rotate

allanlaal, 2021-08-02:
**Describe the bug**

[image]

it only shows a distorted part of the viewfinder. 
landscape works great
reverse landscape also messed up [important for some filming use cases with weird tripod situations]

**To Reproduce**
Steps to reproduce the behavior:

1. use Android 10
2. disable system autorotation (since it seems FreeDCam handles rotation by itself)
3. disable all rotation apps
4. launch FreeDCam
5. rotate phone physically 4x90° to see the difference

**Expected behavior**

rotation works in all 4 positions like anyone would expect

**Smartphone:**
  Device:

Blackview BV6300Pro

Android w kernel 4.14:

[image]

KillerInk, 2021-08-02:
not sure how you are able to rotatet it. but that shouldnt work. freedcam is forced to landscape

allanlaal, 2021-08-02:
my bad, it seems the screen rotation force hack was in effect after stopping the daemon

report invalid, closing

closed 2021-08-02 by allanlaal