KillerInk/FreeDcam #177: Can the camera be turned off when in settings or gallery?

OliverLew, 2021-09-21:
**Describe the bug**
When in settings, the camera is still on (a camera indicator shown in status bar). I think that is quite power consuming, my phone get warm after I browse through settings.

**To Reproduce**
Steps to reproduce the behavior:
I guess, no need

**Expected behavior**
The camera turns off after swiping right (to settings) or left (to gallery).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Oneplus 8
 - OS: Lineage OS 18.1
 - Version 4.3.52

**Additional context**
Add any other context about the problem here.

KillerInk, 2021-09-22:
thats correct and the expected behavior.
settings, camera ui and gallery are fragments attached to pager adapter wich overlay the preview.
normaly you dont spend much time on settings and inbuild gallery.

myself shot most time stars. 
when a shot is done i can check it directly while the camera starts the next capture (phone is remote controlled)
thats the main reason why it is as it is.

your phone should get warm everywhere.
if you are using the preview postprocessing it may caused by that.
the frame processing need some ressources^^

greets

OliverLew, 2021-09-23:
> thats correct and the expected behavior.

I guess that's the cue for me to close the issue :)

> thats the main reason why it is as it is.

I have seen other projects that the program behavior adapts to the personal preferences of the dev, and I respect that. Only that way, fewer user will like and then use the app, no offense meant.

> your phone should get warm everywhere.  if you are using the preview postprocessing it may caused by that.

I have the preview postprocessing disabled.

I guess using the camera and showing the preview alone consumes quite some power, thus the heating. So my point is: fewer power consumption when we are not seeing the preview.

KillerInk, 2021-09-23:
> I guess that's the cue for me to close the issue :)

No^^ a setting could get added to add that behavior. but no idea when it happen. todo list is long.

> I have seen other projects that the program behavior adapts to the personal preferences of the dev, and I respect that. Only that way, fewer user will like and then use the app, no offense meant.

No worry^^ For me this is a hobby. working on it when im motivated. 

> I have the preview postprocessing disabled.
> 
> I guess using the camera and showing the preview alone consumes quite some power, thus the heating. So my point is: fewer power consumption when we are not seeing the preview.

currently checked the power usage on my poco f2 and its not so heavy

[image]