KillerInk/FreeDcam #178: F-Droid build failed

linsui, 2021-09-23:
```
Could not determine the dependencies of task ':app:processReleaseResources'.
> Could not resolve all task dependencies for configuration ':app:releaseCompileClasspath'.
   > Could not resolve project :renderscript-intrinsics-replacement-toolkit:renderscript-toolkit.
     Required by:
         project :app
      > No matching configuration of project :renderscript-intrinsics-replacement-toolkit:renderscript-toolkit was found. The consumer was configured to find an API of a component, preferably optimized for Android, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' but:
          - None of the consumable configurations have attributes.
```

Could you please take a look? Thanks!

KillerInk, 2021-09-25:
do the build bot not support git submodules?

linsui, 2021-09-25:
It supports submodule but need to be enabled. I'll fix it, thanks!

linsui, 2021-09-26:
The build failed with ndk 12. See https://gitlab.com/linsui/fdroiddata/-/jobs/1623457782. Could you please take a look? Thanks!

KillerInk, 2021-09-27:
dont use ndk 12. its no longer compatible with latest gradle. 
with lastet ndk it should build

linsui, 2021-09-27:
I also tested ndk r22b but failed with another error. https://gitlab.com/linsui/fdroiddata/-/jobs/1623484968

KillerInk, 2021-09-27:
think it has something todo with that message:

`[CXX5106] NDK was located by using ndk.dir property. This method is deprecated and will be removed in a future release. Please delete ndk.dir from local.properties and set android.ndkVersion to [22.1.7171670] in all native modules in the project. https://developer.android.com/r/studio-ui/ndk-dir`

latest gradle dont like older ndks

linsui, 2021-09-27:
I usually see that but it won't break the build. Should I test other NDK versions?

KillerInk, 2021-09-27:
i have no idea whats going on. here its working
im building with ndk **21.4.7075529** and android studio **artic fox 2020.3.1 patch2** and **gradle-7.0.2** wich seem to be latest.
os win10x64

linsui, 2021-09-27:
NDK r21e works, thanks! You can add it to build.gradle to ensure the correct NDK is used.

closed 2021-09-27 by linsui