KillerInk/FreeDcam #180: Return feature: take capture via volume buttons

dimka11, 2021-11-28:
In recent version there is such feature but it was in one of previous version

KillerInk, 2021-12-28:
that feature got replaced. someone wanted the ability to change manuals wie vol+/-
you can use now the headset to start a capture.
click long on center button to start a capture. press short to switch to next manual mode.
use vol buttons to de/increase selected manual

dimka11, 2021-12-28:
@KillerInk why just don't add settings for control this feature?

muellerto, 2022-09-07:
What happened to this issue now? Will FreeDCam (like every good camera app) in future be able to take a picture using the volume buttons or not? If there are multiple interests in these buttons ... why not make them configurable?

rubienr, 2022-09-12:
It would be helpful if the volume buttons' action could be made configurable. I would like to use the app for photogrammetry scanning where the volume button is actuated by a servo. Example: [photogrammetry scanner](https://github.com/photogrammetry-scanner)).

KillerInk, 2022-09-13:
its on my todo list. may happen in the next half year.
@rubienr seems a funny project