KillerInk/FreeDcam #182: Support cropped RAW resolutions

epicfilmmaker93, 2021-12-28:
Currently if I select a cropped resolution in Picture Mode and take a photo in 16-bit mode using both the FreeDCam DNG converter and the Android DNG converter the output is full res. If I select other options, the camera just crashes. Cropped resolutions are useful for RAW Video

KillerInk, 2021-12-28:
its currently forced to largest raw_sensor size and freedcam converter. 
it makes no difference to performance aslong it gets not written. 
plans are that it get cropped on dng write out.
its also possible that it get dropped when its not possible to get constant fps

epicfilmmaker93, 2021-12-28:
If you crop it after the capture there would be the same problem as with MotionCam - lower maximum framerate. For example, my phone can record at 120 fps when the image is cropped to 16:9 (tested in video mode) but only 96 fps when I use the full sensor, so please consider cropping before capture

KillerInk, 2021-12-29:
you can not compare video recording with that approach. 
in theory i may be possible to get 120fps with raw frames aslong nothing gets processed or written to sd.
but writing that amount of data is the bottle neck.
now back to cropping. at some point i have to write data to sd. 
if i do the cropping there, then there is no performance lost. it get even faster, because less data get written into the dng.
lets do some math:
lets assume a full frame have a res of 4624x3472 with 16bit = 32.109.056bytes
that get now center cropped to 1920x1080 with 16bit = 4.147.200bytes
so in theory we can write 8x more files.

epicfilmmaker93, 2021-12-29:
> but writing that amount of data is the bottle neck.

Indeed, but please check out my pull request (where I've tagged you) - with MotionCam my device can record 96FPS 3884x1666 for about 3 seconds. Slowing it down to 24FPS would make the video 12 seconds long - enough for a lot of cases. If the RAW frames were cropped before capture, my phone could record 120 FPS and would probably do that for about 2.4 seconds, which when slowed down would also make the video 12 seconds long. My point is that even recording for less than 3 seconds can have great results, and while there are bottlenecks on current devices, they might not be there on the next year's devices - ROG Phone 5s has 16GB of RAM - enough for 10+ seconds of 4K120.

closed 2021-12-30 by epicfilmmaker93