KillerInk/FreeDcam #188: App opens for a split second, then crashes

CodeCubeNeo, 2022-02-22:
**Describe the bug**
You open the app, it shows the controls for a split second, then crashes.

*Smartphone (please complete the following information):**
 - Device: Huawei Mate 9
 - OS: Android9
 - Version: 4.3.52

**Additional context**
the previous version of the application (4.3.51) works fine.
here is the log.txt of the apllication (4.3.52) after it crashes (the error is described at the bottom)

[log.txt](https://github.com/KillerInk/FreeDcam/files/8119848/log.txt)

tmalone14, 2022-02-23:
Me too.
Huawei Mate 9
Android 8
Version 4.3.52

KillerInk, 2022-02-24:
should fix it 7f1ed62389a05c4645b3d3945c5f4eda7c32d293

CodeCubeNeo, 2022-02-24:
Great, thanks. Are there nightly releases or do i have to build the app myself to try this out?

KillerInk, 2022-02-28:
there a no nightly releases. you can build on your own, or wait till a new release gets pushed