KillerInk/FreeDcam #191: Video recording lag (hevc) in latest Ver (4.3.65)

fob014, 2022-03-13:
Video (hevc) recording lag in latest version 4.3.65 
there was no such lag (hevc) in previous version 4.3.52 its all smooth
in hevc codec 
android 11 
cam 2 api

KillerInk, 2022-03-15:
tried it on different devices. im sorry, but i can not reproduce it.

fob014, 2022-03-25:
> tried it on different devices. im sorry, but i can not reproduce it.

ok thanks for reply 

i find out long Sutter mode beyond 1/2 not working 
in previous version it goes up to 32 sec and work fine too

KillerInk, 2022-04-13:
> i find out long Sutter mode beyond 1/2 not working

happen this only with raw or jpeg too?

fob014, 2022-04-13:
> > i find out long Sutter mode beyond 1/2 not working
> 
> happen this only with raw or jpeg too?

In both it doesn't work but in earlier version it work in both jpeg / dng 

I even edit xml file to get long exposure but after that it  does show exposure  up to 8, 16 , 32 eg in app  
But when i press shutter it doesn't work