KillerInk/FreeDcam #192: 108MP not working - Xiaomi Mi10

kylemd, 2022-03-24:
Xiaomi Mi10 umi
February Pixelexperience A12 ROM
FreeDCam 4.3.65 F-Droid

No logs sorry - FreeDCam isn't writing to log.txt!
Not sure if that is a bug or my fault.

Steps to reproduce:
Open cam, settings, run feature detector.
Take photo at 108MP resolution.
Photos take at quad Bayer res (27.1MP)

Is there any chance of getting 108MP working?

KillerInk, 2022-04-11:
xiaomi allows at current gen devices only quad dngs. 
not sure whats wrong with 108mp res for jpeg. as far i know it should work.
maybe its a problem from the custom rom.

Iey4iej3, 2022-12-20:
I am on LOS with the same issue.

This post about OpenCamera might be related: https://forum.xda-developers.com/t/enable-the-5-cams-at-opencamera-including-108mpx-mode.4016279/#post-81252809