KillerInk/FreeDcam #197: Cannot identify the proper settings for Redmi 6

KianaKaslana638, 2022-05-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. Xiaomi PocoF2]
 - OS: [e.g. Android9]
 - Version [e.g. 4.3.8]

**Additional context**
Add any other context about the problem here.

KillerInk, 2022-05-04:
[image]

not sure whats your problem. its working as it should.
cannot realy say if the colors are correct or red blue is switched, or if its the matrix. that stuff you have to find out by yourself.
but bggr looks more correct to me

maybe you are lucky and one is already added that works with your raw
if not you have to create on your own or extract it from chromatix libs
for color matrix read this https://github.com/KillerInk/FreeDcam#how-to-create-a-custom-matrix

KianaKaslana638, 2022-05-04:
FreeDcam is still crashing on my phone. Is there any way to convert that in
dcraw?

KillerInk, 2022-05-05:
You can manually add a dng profiles.xml with that content
<devices>
<device name = "">
<opcodes>
</opcodes>
<filesize size= "0">
<blacklvl>64</blacklvl>
<whitelvl>1023</whitelvl>
<width>4224</width>
<height>3136</height>
<rawtype>6</rawtype>
<colorpattern>bggr</colorpattern>
<rowsize>0</rowsize>
<matrixset>Nexus6</matrixset>
</filesize>
</device>
</devices>