KillerInk/FreeDcam #201: App bricks when alternate camera selected - Pixel 6 Pro

thekojent, 2022-08-30:
**Describe the bug**
Using camera2 api, it is only possible to select Back 0 or Front 1 camera. Selecting other cameras (e.g. Back 2) causes a black screen and loss of UI, so the camera cannot be changed to a working one without reinstalling the app.

**To Reproduce**
Steps to reproduce the behavior:
1. Open FreeDcam
2. Change camera to Back 2

**Expected behavior**
Camera changes to other rear lens

**Smartphone (please complete the following information):**
 - Device: Google Pixel 6 Pro
 - OS: GrapheneOS 
 - Version 4.3.65

KillerInk, 2022-08-31:
hi you builded from source or used latest apk from release?

thekojent, 2022-08-31:
> hi you builded from source or used latest apk from release?

Installed from F-Droid