KillerInk/FreeDcam #203: feature request: an ordinary way to terminate the app?

muellerto, 2022-09-07:
I'm not sure if I'm right but ...

... is there really no better way to terminate the app than moving it into background using the [->] button in the upper right corner and then using the now visible square button of the OS to get it back into foreground and closing it there? When I'm on a day-long mountain hiking tour I must do this clumsy procedure 30 times or more.

Why can't the [->] button be an ordinary  [x] button and terminate the app reliably? I would prefer this.

KillerInk, 2022-09-13:
thats also solved. that problem was freedcam locked the event and did not pass it further.

muellerto, 2022-09-13:
Yeah, but where to get it? Mine is from F-Droid, and it's the latest available there. The /e/-App store has also no other version, 4.3.65 (265).

muellerto, 2022-12-07:
I still don't have any solution for that. If I tap on the [->] button in the upper right corner the app just disappears but is still runs in background.

And the NextGen GUI which can the user switch on makes it even worse:
[image]
How can I terminate the app here? There's no control at all.

I use 4.3.82 from F-Droid.