KillerInk/FreeDcam #204: feature request: picture quality in 5% steps

muellerto, 2022-09-07:
Currently the quality of the resulting JPEG images can be adjusted in 10% steps. I would like to have a setting of 95%, this is what I use in other apps, so I would prefer 5% steps. Even better would be a possibility to set it directly on a numeric value, say 97% or 82%. If someone wants it, why not?

KillerInk, 2022-09-13:
hi normaly i would say, edit the settings file to your needs, but there its not possible.
related code is there

https://github.com/KillerInk/FreeDcam/blob/1a7cb6d6242dce6b378e8c093e0571676552af1c/app/src/main/java/freed/cam/apis/camera2/parameters/modes/JpegQualityModeApi2.java#L23-L26

feel free to create a pr

muellerto, 2022-09-13:
I felt free to make this issue here.

closed 2022-12-14 by muellerto