KillerInk/FreeDcam #206: Hide all HUD elements

nomeata, 2022-10-20:
**Describe the bug**

I am (ab)-using the app to turn my eInk tablet into an external HDMI monitor (the HDMI input appears as an Android camera it seems), and FreeDcam seems to be the app that gets it most right (others have trouble with the orientation).

But for that use case, I of course don’t need any of the controls on the screen.

Also the scrolling lable below AE is a problem, as any change causes the eInk display to do work.

Would a mode where the camera is just showing the preview image be feasible?

KillerInk, 2022-10-20:
hi
freedcam dont know if its streaming external. its a screen clone what get send.
its not a big problem to add that. adding a switch to settings is enough when you dont need any control^^
but im not sure if that would realy fix your refresh problem on the eink.
the camera preview itself get refreshed more times a second and that get send to hdmi wich should trigger a refresh of the eink.
pls correct me if im wrong.

nomeata, 2022-10-20:
Well, hiding the AE label will fix the problem of unnecessary scrolling.

The refreshing of the actual screen from the HDMI is of course out of FreeDcam’s control and unavoidable, but that’s ok.

In fact, what I should probably really do is clone the code and simply remove all functionality besides “display the camera preview full-screen” and have a dedicated app… maybe once I figure out how to build this (on NixOS, which is always a bit more tricky).