KillerInk/FreeDcam #212: Instantly crashing on LeMax2

stefan-najdovski, 2023-01-16:
**Describe the bug**
Installed the Latest Oct. 2022 version from F-Droid, upon opening the app crashes instantly

Logs:
```
4874-14907 AndroidRuntime          troop.com.freedcam                   E  FATAL EXCEPTION: CameraApiManager
                                                                                                    Process: troop.com.freedcam, PID: 14874
                                                                                                    java.lang.ClassCastException: Cannot cast freed.settings.mode.SettingMode to freed.settings.mode.ApiBooleanSettingMode
                                                                                                    	at java.lang.Class.cast(Class.java:2479)
                                                                                                    	at freed.settings.SettingsManager.get(SettingsManager.java:120)
                                                                                                    	at freed.cam.apis.featuredetector.AbstractFeatureDetectorTask.checkCameraID(AbstractFeatureDetectorTask.java:66)
                                                                                                    	at freed.cam.apis.featuredetector.Camera2FeatureDetectorTask.checkCameraID(Camera2FeatureDetectorTask.java:178)
                                                                                                    	at freed.cam.apis.featuredetector.AbstractFeatureDetectorTask.detect(AbstractFeatureDetectorTask.java:38)
                                                                                                    	at freed.cam.apis.featuredetector.Camera2FeatureDetectorTask.detect(Camera2FeatureDetectorTask.java:81)
                                                                                                    	at freed.cam.apis.featuredetector.CameraFeatureDetector.detectFeatures(CameraFeatureDetector.java:39)
                                                                                                    	at freed.cam.apis.CameraApiManager.loadFeatureDetector(CameraApiManager.java:81)
                                                                                                    	at freed.cam.apis.CameraApiManager.access$400(CameraApiManager.java:30)
                                                                                                    	at freed.cam.apis.CameraApiManager$1.run(CameraApiManager.java:126)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:201)
                                                                                                    	at android.os.Looper.loop(Looper.java:288)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
```

Full logs:
```
2023-01-16 13:52:56.306 15010-15010 GraphicsEnvironment     troop.com.freedcam                   V  ANGLE Developer option for 'troop.com.freedcam' set to: 'default'
2023-01-16 13:52:56.307 15010-15010 GraphicsEnvironment     troop.com.freedcam                   V  ANGLE GameManagerService for troop.com.freedcam: false
2023-01-16 13:52:56.307 15010-15010 GraphicsEnvironment     troop.com.freedcam                   V  Neither updatable production driver nor prerelease driver is supported.
2023-01-16 13:52:56.310 15010-15010 NetworkSecurityConfig   troop.com.freedcam                   D  No Network Security Config specified, using platform default
2023-01-16 13:52:56.310 15010-15010 NetworkSecurityConfig   troop.com.freedcam                   D  No Network Security Config specified, using platform default
2023-01-16 13:52:56.327 15010-15037 native                  troop.com.freedcam                   D  Found attached 0
2023-01-16 13:52:56.327 15010-15037 native                  troop.com.freedcam                   D  Calling class is: java.lang.Class
2023-01-16 13:52:56.328 15010-15038 native                  troop.com.freedcam                   D  Found attached 0
2023-01-16 13:52:56.328 15010-15038 native                  troop.com.freedcam                   D  Calling class is: java.lang.Class
2023-01-16 13:52:56.337 15010-15040 GraphicsEnv             troop.com.freedcam                   E  Failed to get gpu service
2023-01-16 13:52:56.392 15010-15010 ActivityAbstract        troop.com.freedcam                   D  onCreate
2023-01-16 13:52:56.395 15010-15010 SettingsManager         troop.com.freedcam                   D  load Settings
2023-01-16 13:52:56.678 15010-15010 XmlParserWriter         troop.com.freedcam                   D  /storage/emulated/0/Android/data/troop.com.freedcam/files/dngprofiles.xml exists:false
2023-01-16 13:52:56.678 15010-15010 ActivityAbstract        troop.com.freedcam                   D  onCreatePermissionGranted
2023-01-16 13:52:56.680 15010-15010 ActivityAbstract        troop.com.freedcam                   D  initOnCreate()
2023-01-16 13:52:56.680 15010-15010 ActivityFreeDcamMain    troop.com.freedcam                   D  onCreate: 
2023-01-16 13:52:56.682 15010-15010 SecureCamera            troop.com.freedcam                   D  onCreate
2023-01-16 13:52:56.682 15010-15010 CameraApiManager        troop.com.freedcam                   D  Create camera BackgroundHandler
2023-01-16 13:52:56.682 15010-15010 BackgroundHandlerThread troop.com.freedcam                   D  startBackgroundThreadCameraApiManager
2023-01-16 13:52:56.687 15010-15010 SecureCamera            troop.com.freedcam                   D  onResume
2023-01-16 13:52:56.687 15010-15010 ActivityFreeDcamMain    troop.com.freedcam                   D  onResumeTasks() 
2023-01-16 13:52:56.691 15010-15010 FileListController      troop.com.freedcam                   D  needStorageAccessFrameWork true
2023-01-16 13:52:56.691 15010-15010 FileApiStorageDetector  troop.com.freedcam                   D  getStorageDirectory().listFiles() is null!
2023-01-16 13:52:56.697 15010-15010 Compatibil...geReporter troop.com.freedcam                   D  Compat change id reported: 171228096; UID 10148; state: ENABLED
2023-01-16 13:52:56.700 15010-15010 CameraApiManager        troop.com.freedcam                   D  addEventListner ThemeSampleMainFragment
2023-01-16 13:52:56.700 15010-15010 BaseEventHandler        troop.com.freedcam                   D  set EventListner : ThemeSampleMainFragment
2023-01-16 13:52:56.700 15010-15010 BaseEventHandler        troop.com.freedcam                   D  set EventListner : ThemeSampleMainFragment
2023-01-16 13:52:56.700 15010-15010 ThemeSampleMainFragment troop.com.freedcam                   D  onViewCreated
2023-01-16 13:52:56.702 15010-15010 CameraApiManager        troop.com.freedcam                   D  onResume camera null:true PreviewSurfaceRdy:false cameraIsOpen:false
2023-01-16 13:52:56.702 15010-15010 CameraApiManager        troop.com.freedcam                   D  changePreviewPostProcessing()
2023-01-16 13:52:56.702 15010-15010 ThemeSampleMainFragment troop.com.freedcam                   D  changePreviewPostProcessing
2023-01-16 13:52:56.702 15010-15043 CameraApiManager        troop.com.freedcam                   D  BackgroundHandler is null: false features detected: false app version changed: true
2023-01-16 13:52:56.702 15010-15043 CameraApiManager        troop.com.freedcam                   D  load featuredetector
2023-01-16 13:52:56.703 15010-15043 CameraApiManager        troop.com.freedcam                   D  Start FeatureDetector
2023-01-16 13:52:56.703 15010-15010 PreviewController       troop.com.freedcam                   D  init preview off
2023-01-16 13:52:56.703 15010-15043 CameraFeatureDetector   troop.com.freedcam                   D  CameraFeatureRunner process
2023-01-16 13:52:56.704 15010-15010 ThemeSampleMainFragment troop.com.freedcam                   D  onResume
2023-01-16 13:52:56.705 15010-15010 OrientationManager      troop.com.freedcam                   D  start
2023-01-16 13:52:56.738 15010-15043 MediaCodecInfoParser    troop.com.freedcam                   D  MediaCodecInfo Name:c2.android.aac.encoder hwaccel:
[... cut ...]
2023-01-16 13:52:56.952 15010-15010 SecureCamera            troop.com.freedcam                   D  onPause
2023-01-16 13:52:56.975 15010-15010 ActivityFreeDcamMain    troop.com.freedcam                   D  onPauseTasks() 
2023-01-16 13:52:56.975 15010-15010 OrientationManager      troop.com.freedcam                   D  stop
2023-01-16 13:52:56.975 15010-15043 Process                 troop.com.freedcam                   I  Sending signal. PID: 15010 SIG: 9

```

**Smartphone (please complete the following information):**
 - Device: LeMax 2
 - OS: Android 13

stefan-najdovski, 2023-01-16:
Looks like its camera 2 api device specific issue, will report after more debuging

KillerInk, 2023-01-16:
looks like its not supporting camera2 also not in legacy mode. but it seems to report its supported.
on my good old o3d and evo3d its working. only camera1 get detected.
on g3(wich is legacy camera2) it detects camera2 and starts up.

stefan-najdovski, 2023-01-16:
> looks like its not supporting camera2 also not in legacy mode. but it seems to report its supported. on my good old o3d and evo3d its working. only camera1 get detected. on g3(wich is legacy camera2) it detects camera2 and starts up.

Talked with the developer of the ROM he said he is aware of the issue that camera 2 api has issue with Android 13 on this device. Legacy Mode works only with Open Camera, so I would asume its not the FreeDcam fault.

stefan-najdovski, 2023-01-16:
Works fine on A11

closed 2023-01-16 by stefan-najdovski