KillerInk/FreeDcam #218: build error: Cannot invoke "String.startsWith(String)" because "versionOutput" is null

FlashlightET, 2023-10-04:
**Describe the bug**
Simply trying to build i get "Cannot invoke "String.startsWith(String)" because "versionOutput" is null"

**To Reproduce**
Steps to reproduce the behavior:
1. Build project
2. See error

**Expected behavior**
App build continues on

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy A21 (this isnt relevant though)
 - OS: Android 10
 - Version Latest

**Additional context**
Add any other context about the problem here.

KillerInk, 2023-10-05:
hi
cant reproduce it. building fine here.
when i have to guess, from that one line of the build log, i would say you dont have set a version.
and your problem is this line

https://github.com/KillerInk/FreeDcam/blob/60682b2a086988f291725f8c9866a41350abda91/app/build.gradle#L105

KillerInk, 2023-10-05:
and here is a sample how to extend max iso on camera2

https://github.com/KillerInk/FreeDcam/blob/60682b2a086988f291725f8c9866a41350abda91/app/src/main/res/raw/supported_devices.xml#L4384-L4393