KillerInk/FreeDcam #219: On Pixel 4 XL, Face detection not working. Switch to tele lens got black screen/app freezed. Tap to focus doesn't have a metering mode; and also missing Manual Brightness and Contrast in UI icons

Puxirepublic, 2023-12-22:
**Describe the bug**
In settings Face detection whether sets Full or Simple, nothing happened on preview UI when there is a clear face around.
Switch to tele lens got black screen and app freezed, only resets app to get back normal again; uses zoom to switch to tele lens works fine.
Tap to focus doesn't have a metering mode feature; also missing Manual Brightness and Contrast features on preview camera UI icons.

 - Device: Pixel 4 XL
 - OS: Android 13
 - v4.3.8

KillerInk, 2024-02-02:
About facedetection:
Tried it on my good old nex5 and pocof2 and I get there a red border around my face.
No idea what's wrong with the pix4

About telelens:
Ignore that cam. Freedcam brute force the camera ids and look wich one return something. But that not always mean that camera works. But it's the only way to get the cams that are locked by oems.
You can also remove that camera Id from the settings and it disappear on next start.

Camera 2 by don't support brightness and contrast by default.
Some qcom socs support it.
Exposure metering gets always set where the focus spot is. I have no plans to extend it.

Puxirepublic, 2024-02-04:
Thanks for the details. On switching to tele lens, it's not a big deal to me, since I can use manual zoom to activate the tele lens, but then I find a new issue with zoom to the tele lens, it completely ignores the settings - for example, I set the noise off, edge full, in tele lens shot, it's still using default denoise and edge settings.

About face detection, on Open Camera, it works fine with Pixel 4 XL, maybe it's a useful info.