KillerInk/FreeDcam #220: Redmi 4A DNG shot on versions after 4.1beta6 open cropped in Photoshop

JonasHanway, 2023-12-29:
My Redmi 4A supports raws with 4208x3120 resolution, but for some reason DNGs made by latter versions of FreeDcam get cropped to 4192x3104 when opened in Photoshop's Camera Raw. I tried changing app's settings, DNG profile settings, but nothing helped. Only reinstalling old 4.1beta6 gave me full sized pictures again. Starting from 4.1beta9 and later the result is cropped. Any chance you could fix it somehow in future updates?

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Xiaomi Redmi 4A
 - OS: Android 7.1.2
 - Version 4.3.8

KillerInk, 2024-01-24:
Hi sorry for late response.
That is correct. 
Dng specs tells that about:

> Raw images often store extra pixels around the edges of the final image. These extra pixels help prevent interpolation artifacts near the edges of the final image.

That means that the full image is stored. but the image processor(rawtherappee/photoshop/...) use that 8 rows/lines on top/bottom left/right for pre demosaicing to get no artrifacts in the edges^^

greets

JonasHanway, 2024-01-24:
But non-cropped DNGs look completely fine and have no artifacts on edges or other problems, which makes this crop absolutely pointless. I'll just have to use the old version if further updates have no option of disabling this crop.

MaferCJ, 2024-02-20:
You can remove the crop tags with exiftool (on Android you can use Termux)

https://github.com/darktable-org/darktable/issues/11949#issuecomment-1141806518