KillerInk/FreeDcam #233: JPG is taken from JPEG stream or manually from RAW?

Uzver123, 2024-09-29:
I wonder how JPG is taken using phone JPEG stream or converted from RAW data internally by FreeDCam?

KillerInk, 2024-09-30:
its using the jpeg stream. freedcam itself dont have a demosaic algo. what you get in the viewer is from libraw.
greets

Uzver123, 2024-09-30:
> its using the jpeg stream. freedcam itself dont have a demosaic algo. what you get in the viewer is from libraw.

Then how FreeDcam has different options for sharpening from standard off/fast/high/zsl that offered in Camera2API?

[image]

Uzver123, 2024-09-30:
I have phone OnePlus 8 Pro if i take photo in OpenCamera with sharpening off it shows good on preview, but when i take photo it somehow switches sharpening to max there is no difference between sharpening off and high settings for JPG.

In freeDcam if i set sharpening to zero it really is zero, i wonder what's the difference and if understanding this difference can help me locate the way OnePlus alters JPG so i can disable it (i have root).

KillerInk, 2024-10-01:
Don't know how good you coding skills are. But you have two possibilities
First mod the camera apk you need.
Second if it's a qcom socs, mod libchromatix.lib that the jpeg stream don't use sharpening.
When I played with it there was only one. But these days I think, is one for every mode.

Uzver123, 2025-01-19:
> its using the jpeg stream. freedcam itself dont have a demosaic algo. what you get in the viewer is from libraw.
> greets

By the way FreeDcam has re-mosaic algo? i know someone who is used it for forcing re-mosaic for camera sensor that had no HW-remosaic in full resolution mode, but when i tried that did not work.

KillerInk, 2025-03-05:
it have. QUADBAYER_16BIT 7. set that inside the dng profile and set the correct rawsize and it should work.

Uzver123, 2025-03-05:
> it have. QUADBAYER_16BIT 7. set that inside the dng profile and set the correct rawsize and it should work.

Do you have detail walk trough how to set up?

KillerInk, 2025-03-07:
1. give freedcam sd permission.
2. open freedcam and set sdcard to external and choose a folder where the "raw" binary data should get stored.
3. select "raw" as picture format inside freedcam.
4. select the full resolution and capture a image.
5. close freedcam and open freedviewer and go to the folder where the "raw" file is.
6. on the bottom right press the "..." and choose raw to dng, select the "raw"file
7. enter there all the needed information and save the dng profile. there you can also choose QUADBAYER_16BIT 
8. now you should be able to capture fullsizedngs dequaded